MuonPi/muondetector-old-qt5-backend #24: Failed to open port /dev/ttyAMA0, errors

alexschultze, 2020-12-28:
Remark: This is a "dry" run without any actual hardware present on a RPI 1 B+.

```
pi@raspberrypi:~ $ muondetector-daemon
Warning: No 'ublox_device' setting in configuration file. Will guess...
Info: detected "/dev/ttyAMA0" as most probable candidate
Critical: eeprom device NOT present!
Info: Found HW version 2 in eeprom
Warning: QObject::connect: invalid null parameter
Warning: LM75 device NOT present!
Warning: ADS1115 device NOT present!
Critical: MCP4728 device NOT present!
Critical: PCA9536 device NOT present!
Failed to open port /dev/ttyAMA0, error: Permission denied
trying again in 5000 seconds
Segmentation fault
```
By default Serial Port Console is enabled. Should be mentioned somewere in the setup procedure for Pi Os.
Disabling the Serial Port Login Console according to this source:
https://www.raspberrypi.org/forums/viewtopic.php?t=180254

```
Warning: No 'ublox_device' setting in configuration file. Will guess...
Info: detected "/dev/ttyAMA0" as most probable candidate
Critical: eeprom device NOT present!
Info: Found HW version 2 in eeprom
Warning: QObject::connect: invalid null parameter
Warning: LM75 device NOT present!
Warning: ADS1115 device NOT present!
Critical: MCP4728 device NOT present!
Critical: PCA9536 device NOT present!
Critical: "Unable to start the server: The bound address is already in use.\n"
Failed to open port /dev/ttyAMA0, error: Permission error while locking the device
trying again in 5000 seconds
Segmentation fault
```
User is part of dialout:
```
pi@raspberrypi:~ $ groups
pi adm dialout cdrom sudo audio video plugdev games users input netdev lpadmin gpio i2c spi
```

alexschultze, 2020-12-28:
Maybe change default to /dev/serial0 ?

https://raspberrypi.stackexchange.com/questions/49748/fatal-cannot-open-dev-ttyama0-permission-denied

marvin5300, 2020-12-28:
Thank you for this test. This code here:

`        qWarning() << "No 'ublox_device' setting in configuration file. Will guess...";`
`        QDir directory("/dev","*",QDir::Name, QDir::System);`
`        QStringList serialports = directory.entryList(QStringList({"ttyS0","ttyAMA0"}));`
`        if (!serialports.empty()){`
`            gpsdevname=QString("/dev/"+serialports.last());`
`            qInfo() << "detected" << gpsdevname << "as most probable candidate";`
`        }else{`
`            qCritical() << "no device selected, will not connect to GNSS module" << endl;`
`        }`
was intended to automatically select 'ttyAMA0' or 'ttyS0' for raspberry pi 2 and raspberry pi 3. My guess is that in a newer raspi version 'serial0' is the correct name. There is a config file in '/etc/muondetector/muondetector.conf' where this serialport can be set manually.

Another point is that we assumed the daemon would run as service anyway so calling 'muondetector-daemon' directly may be broken. We have to look into this.

alexschultze, 2020-12-28:
I updated to /dev/serial0 in the given config file and it starts without problems now, even called directly.

More Raspi Version Info:
```
pi@raspberrypi:~ $ cat /etc/debian_version
10.6
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
```

closed 2021-01-29 by dtreffenstaedt