MuonPi/muondetector-old-qt5-backend #47: GUI keeps disconnecting and/or crashing

kimbou, 2021-05-26:
**Describe the bug**
When running the gui for multiple days straight it sometimes freezes, randomly disconnects or closes (crashes) completely. 
The daemon seems not to be affectet and i can reconnect if it happens. 
This problem mostly occured when i left it running for more that 24h

marvin5300, 2021-05-26:
Which system is the GUI running on? Raspberry Pi or other? Does the system go to energy saving mode?

kimbou, 2021-05-27:
> Which system is the GUI running on? Raspberry Pi or other? Does the system go to energy saving mode?

I am using a Raspberry Pi, no it does not

dtreffenstaedt, 2021-05-28:
I have just started a long term test to see if I can reproduce the error.

In the meantime, could you please provide us extra details about your setup?

- Raspberry Pi model
- operating system name and version
- Does the GUI run on the same raspberry pi as the daemon?
- Is the Raspberry Pi connected via WiFi or cable?

Any additional information you have that you think might be helpful is welcome.

dtreffenstaedt, 2021-05-30:
After a test over two days I can not reproduce the issue on my setup. If anyone else could confirm and provide more information, we could get closer to the issue.

kimbou, 2021-05-31:
> I have just started a long term test to see if I can reproduce the error.
>
> In the meantime, could you please provide us extra details about your setup?
>
>     * Raspberry Pi model
>
>     * operating [...]

 I am using a Raspberry Pi3 Model B, the operating system is Raspbian GNU/Linux 10 and both the daemon and the gui run on the same rasbpi. 
It was connected to WiFi via cable when it happened the first time, now its not anymore

CodingBobby, 2022-03-04:
I have experienced a similar behaviour of the GUI where it disconnects without indication, or rather, doesn't reconnect automatically. For me, it doesn't happen randomly and not on the same RPi where the daemon is running, so I don't know if this issue is related in any way.

I have installed the GUI on a Mac (Mojave 10.14.6) and when awaking after going into sleep mode, the GUI doesn't receive any further updates. The "Time Pulse" label is red and no more event pulses are shown. The connection status still says "connected" and the statuses of the i2c devices say "online". The GUI is basically halted at the moment of going into sleep mode. Buttons are still clickable, but don't show any effect when the actual Pi would be involved.

The bug can be solved easily by disconnecting and then connecting again.

Testing a few things, I found that the same behaviour occurs when loosing WiFi signal (or disabling WLAN temporarily) for several minutes. When the signal outage lasts for a few seconds only, the GUI simply continues to work. Maybe a similar thing happened to @kimbou ? This would mean that even though the GUI could still communicate with the daemon, a loss of connection to the internet (or MQTT server) breaks it.

kimbou, 2022-03-11:
@CodingBobby  the RasPi in question is connected via ethernet and there have been no outages reported by the university but it could be possible that i have lost connection to the internet. It always occured when i wasn't around to watch

hangeza, 2022-04-03:
An automatic timeout/disconnect functionality was implemented for the GUI with commit dece6d355ceaf2ac2565899fea7591f7d0f77772. Now, at least the GUI is not in a floating or stale state, when the traffic from the daemon interrupts, but rather cleanly put into disconnected state. The timeout was chosen to be 20 seconds which still has to be tested thoroughly and replaced by a better value. However, this would solve the problem obeserved by @CodingBobby. Therefore this issue can be closed, i guess.