MuonPi/muondetector-old-qt5-backend #62: Backward compatibility with armv6 (Raspberry Pi Zero)

marvin5300, 2022-01-05:
Need to add CMakeLists.txt options for armv6
Maybe need to compile daemon for armv6?

lnies, 2022-01-06:
Currently debugging on Raspberry Py Zero v1.1 WH. 

- Cloning main #51 and not changing anything, building package. Only GUI is being built. GUI seems to start without problems, couldn't test functionality without daemon though yet. Daemon is not built because of 
https://github.com/MuonPi/muondetector/blob/81eefc3bafef4e01d1961e678da4d55d1a156b03/CMakeLists.txt#L18-L21
- This part here probably needs also attention:
https://github.com/MuonPi/muondetector/blob/81eefc3bafef4e01d1961e678da4d55d1a156b03/CMakeLists.txt#L56-L61
- Changing this to `if ("${CMAKE_SYSTEM_PROCESSOR}" MATCHES "^(armv7|armv6)")` as recommended by Daniel, CMake executes w/o problems, but runs into following problem (on my Pi):
```
[ 43%] Building CXX object CMakeFiles/muondetector-daemon.dir/daemon/src/pigpiodhandler.cpp.o
/home/pi/muondetector_daemon_binary/muondetector/daemon/src/pigpiodhandler.cpp:13:10: fatal error: pigpiod_if2.h: No such file or directory
   13 | #include <pigpiod_if2.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
````
- This could be system related problem where some libraries are mission. `pigpiod -v` yields `79`, however `sudo apt update && sudo apt install pigpiod` installs additional packages: `libpigpio-dev libpigpiod-if-dev libpigpiod-if1 pigpio pigpio-tools`. After, I browse for additional packages and also install `libpigpiod-if2-1

marvin5300, 2022-01-06:
Looks like pigpiod include directories have to be installed manually before compiling. Nothing unusual, maybe add it to the list of dependencies when building yourself?

lnies, 2022-01-16:
Following the steps I outlined in my previous comment, the daemon and the GUI compile under Bullseye on ARMv6 architecture. The compiled daemon can be downloaded here: https://archive.muonpi.org/2.0.2/raspberry_pi_zero_arrmv6/bullseye/muondetector-daemon_2.0.2-hotfix1_armhf.deb.

closed 2022-02-02 by lnies

gon0, 2022-12-30:
Hello everyone,

I have two Raspberry Pi Zero W. I just installed the newest bullseye image and performed `sudo apt update && sudo apt upgrade` as explained here https://wiki.muonpi.org/index.php?title=Raspberry_Pi_Setup .

At the "interfaces" part I enabled Serial and I2C as described. When I "finished" the config, I received this error:

```
pi@raspberrypi:~ $ sudo raspi-config
Created symlink /etc/systemd/system/multi-user.target.wants/vncserver-x11-serviced.service → /lib/systemd/system/vncserver-x11-serviced.service.
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.15.61+

```
although the config displayed:
[image]

This happened on both Pi Zero W.

What could have gone wrong? Does it make sense to go on with the installation of the muonPis?