MuonPi/muondetector-old-qt5-backend #64: mqtt login breaks and daemon can't be started

CodingBobby, 2022-01-14:
I'm not able to connect to the muonpi-network because the login-process breaks.

**Steps to reproduce the behaviour:**
I followed the provided guide (https://wiki.muonpi.org/index.php?title=Raspberry_Pi_Setup#Guided_Setup) and set up a fresh installation of the `muondetector-lite` image.
When starting the Raspberry for the first time and connecting via `ssh`, the `mqtt` login process breaks right after it started.

**This is the output I get:**
```
Welcome to your muondetector.
For this first setup, please set a new password for the pi user:
New password:
Retype new password:
passwd: password updated successfully
Starting and stopping muondetector-daemon. (hack: https://github.com/MuonPi/muondetector/issues/38)
Testing internet connection...
Established.
Now enter your mqtt credentials for the muonpi network:
Illegal instruction
```

**I tried several things:**
1. run `muondetector-login`, the output is still "`Illegal instruction`"
2. run `sudo runuser -u muonuser -- muondetector-login`; it is asked to type in the "`password for muonuser`"; neither the custom password for `pi` nor "raspberry" or an empty one works.
3. run `sudo runuser -u pi -- muondetector-login` as `pi` was the user I changed the password for; output is again "`Illegal instruction`"
4. running `muondetector-daemon` directly throws the error "`error while loading shared libraries: libpigpiod_if2.so: cannot open shared object file: No such file or directory`"
5. first `sudo apt update && sudo apt upgrade`, then the above steps again; all outputs remain the same.
6. put mqtt credentials into `/etc/muondetector/muondetector.conf` and restarted daemon with `sudo systemctl restart muondetector-daemon.service`; when checking `systemctl status muondetector-daemon.service` it shows:

```
● muondetector-daemon.service - muondetector-daemon - Daemon for custom muondetector board
   Loaded: loaded (/lib/systemd/system/muondetector-daemon.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2022-01-14 22:21:20 CET; 54min ago
  Process: 2989 ExecStart=/usr/bin/muondetector-daemon -c (code=exited, status=127)
 Main PID: 2989 (code=exited, status=127)

Jan 14 22:21:20 muondetector systemd[1]: muondetector-daemon.service: Service RestartSec=100ms expired, scheduling
Jan 14 22:21:20 muondetector systemd[1]: muondetector-daemon.service: Scheduled restart job, restart counter is at
Jan 14 22:21:20 muondetector systemd[1]: Stopped muondetector-daemon - Daemon for custom muondetector board.
Jan 14 22:21:20 muondetector systemd[1]: muondetector-daemon.service: Start request repeated too quickly.
Jan 14 22:21:20 muondetector systemd[1]: muondetector-daemon.service: Failed with result 'exit-code'.
Jan 14 22:21:20 muondetector systemd[1]: Failed to start muondetector-daemon - Daemon for custom muondetector board
```

If it matters, I have fully set up the hardware (RasPi Zero and MuonPi Lite Board). The LEDs for `VCC` and `+5V` are on and `Timepulse` blinks in an interval, the others are off.

Thanks for your help in advance!

hangeza, 2022-01-15:
Dear CodingBobby,

We learned just recently that the zero utilizes the armv6 architecture. Currently, all muonpi binary packages are only compiled for the standard armv7 architecture  and do not foresee armv6. However, the CMakeList files of the daemon and gui projects have been updated to recognize armv6 in the meantime in the dev branch. So, I propose the following process to obtain the correct packages yourself until we are able to update the muonpi repos with armv6 binaries:

1) uninstall  the installed packages "muondetector-daemon" and (if applicable) "muondetector-gui" via apt
2) clone the muondetector-repo into a local folder
3) install all prerequisite packages for building (see "Cheat-Sheet Copy&Paste" on git-repo overview page)
4) build the project as described on the git overview page and install the binary (-ies) via apt install

In case you are using the Lite image, no graphical dependencies are available. In this case you might want to suppress the compilation of the GUI with setting the flag "MUONDETECTOR_BUILD_GUI" in the main CMakeLists.txt to OFF.

The binary packages will be made available in the apt repo asap. Since only a small number of developers currently have a Zero, the testing might take a bit, though.

lnies, 2022-01-16:
I am working on compiling a package for the armv6 architecture on the Pi Zero, will let you guys know when it is available. Compiling on the Zero takes its sweet time... Will report debugging in #62.

CodingBobby, 2022-01-16:
Right, thanks for your quick help. I didn't know this issue could be caused by the architecture.

I followed your steps, including those mentioned in #62 (modifying `CMakeLists.txt`, installing additional dependencies), and it compiled well into a `muondetector-daemon*.deb`. After installation, the daemon couldn't be started still.

- (re)starting the daemon service and checking its status shows:
```
● muondetector-daemon.service - muondetector-daemon - Daemon for custom muondetector board
   Loaded: loaded (/lib/systemd/system/muondetector-daemon.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2022-01-15 22:40:49 CET; 1s ago
  Process: 10991 ExecStart=/usr/bin/muondetector-daemon -c (code=killed, signal=ILL)
 Main PID: 10991 (code=killed, signal=ILL)

Jan 15 22:40:49 muondetector systemd[1]: muondetector-daemon.service: Service RestartSec=100ms expired, scheduling
Jan 15 22:40:49 muondetector systemd[1]: muondetector-daemon.service: Scheduled restart job, restart counter is at
Jan 15 22:40:49 muondetector systemd[1]: Stopped muondetector-daemon - Daemon for custom muondetector board.
Jan 15 22:40:49 muondetector systemd[1]: muondetector-daemon.service: Start request repeated too quickly.
Jan 15 22:40:49 muondetector systemd[1]: muondetector-daemon.service: Failed with result 'signal'.
Jan 15 22:40:49 muondetector systemd[1]: Failed to start muondetector-daemon - Daemon for custom muondetector board
```
- running `muondetector-daemon` throws the error "`illegal hardware instruction  muondetector-daemon`"
- logging in with the mqtt credentials still doesn't work and behaves like I described initially

Cannot say if I have done anything wrong with the compilation, it took quite a while and I didn't follow the logs.

lnies, 2022-01-16:
It's still compiling on my side, I'll let you know when the package is uploaded. Which OS are you running? For me, I am building on Debian Bullseye, a Raspberry Pi OS Light installation. I had to download a lot of libraries for the dependencies, especially to build the GUI. But if you say you didn't get any compiling errors, then that should be fine.

hangeza, 2022-01-16:
Successful compilation on a system should always yield executable code on this very system. I would guess, that the deb package was not installed properly. Make sure, that you install the package from file and not the equally named package from the repo again. Furthermore you might want to tell apt to force the package installation even if the package to be installed is not newer than the currently installed one. To do so, try

sudo apt install ./muondetector-daemon... --reinstall

If the problem still persists, we have to search somewhere else...

lnies, 2022-01-16:
I agree with @hangeza, I would first uninstall the daemon with apt and even remove the repository and the key from the package manager altogether. Afterward, you can try to install your compilation again, and if it still doesn't work, try to download my compilation here: https://archive.muonpi.org/2.0.2/raspberry_pi_zero_arrmv6/bullseye/muondetector-daemon_2.0.2-hotfix1_armhf.deb.

CodingBobby, 2022-01-16:
I'm running the OS from the lite-image you provide here https://archive.muonpi.org/image/latest/. After finding the errors, I stayed on that system. Should I try a clean manual installation on a raw standard OS? Could `muondetector-login` interfere with the daemon? I don't think any of that was touched during my modifications.

I removed the existing installation of the daemon and made sure to install the actual file of the compiled package using your suggested command as well as doing it with `sudo dpkg -i ./muondetector-daemon*.deb` which doesn't check against other package sources. Both methods resulted in the behaviour I described.

@lnies thanks for your compilation, will try it out soon.

CodingBobby, 2022-01-18:
Installation of the package you built was successful, and I was even able to log in with my mqtt credentials. Though, the daemon seems to have other problems now, as no detection-data is written, and I cannot interact with the detector to change settings.

When (re)starting the daemon, this is the status:
```
● muondetector-daemon.service - muondetector-daemon - Daemon for custom muondetector board
   Loaded: loaded (/lib/systemd/system/muondetector-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2022-01-18 11:15:37 CET; 7s ago
 Main PID: 1712 (muondetector-da)
    Tasks: 9 (limit: 877)
   CGroup: /system.slice/muondetector-daemon.service
           └─1712 /usr/bin/muondetector-daemon -c

Jan 18 11:15:37 muondetector systemd[1]: Started muondetector-daemon - Daemon for custom muondetector board.
Jan 18 11:15:38 muondetector muondetector-daemon[1712]: Info: MuonPi Muondetector Daemon  "V2.0.2-f3a95c6" "(build Sun Jan 16 20:05:46 2022)"
Jan 18 11:15:38 muondetector muondetector-daemon[1712]: Info: EEP unique ID: "0x294100556d82"
Jan 18 11:15:38 muondetector muondetector-daemon[1712]: Info: Found HW version 3 in eeprom
Jan 18 11:15:38 muondetector muondetector-daemon[1712]: Warning: QObject::connect: invalid null parameter
Jan 18 11:15:38 muondetector muondetector-daemon[1712]: Critical: PCA9536 device NOT present!
Jan 18 11:15:40 muondetector muondetector-daemon[1712]: Info: Connected to mqtt.
Jan 18 11:15:40 muondetector muondetector-daemon[1712]: Info: MQTT (re)connected
```
I am worried about the warning and the critical error, what do they mean? The hardware seems to work and detect muons already, as all LEDs light up and blink as they should.

I have (apparently) successfully connected the GUI from another computer, but all data-fields only show placeholders and none of the buttons work. After any interaction, I find loads of this error message in the daemon status-log:
```
Jan 18 12:12:52 muondetector muondetector-daemon[1740]: Warning: QIODevice::write (QFile, "/var/muondetector/3d5a6b5cae9374a06e606329abe799e7243519929048d7715365fece/notUploadedFiles/log_2022-01-18_10-17-52.dat"): device not open
```
Is this a permission-caused issue? I checked, and the user `pi` has ownership rights for the `/var/muondetector`-folder.

Running commands like `muondetector-daemon --showout` yield this output:
```
Info: MuonPi Muondetector Daemon  "V2.0.2-f3a95c6" "(build Sun Jan 16 20:05:46 2022)"
Info: EEP unique ID: "0x294100556d82"
Info: Found HW version 3 in eeprom
Warning: QObject::connect: invalid null parameter
muondetector-daemon: relocation error: muondetector-daemon: symbol _ZN20QTextStreamFunctions4endlER11QTextStream version Qt_5 not defined in file libQt5Core.so.5 with link time reference
```

Thank you guys for taking your time to look at my heap of difficulties.

hangeza, 2022-01-19:
It still looks like an incompatible ABI probably coming from the fact that the binary was linked against a different version of Qt than what is present on the image (at least i think is true for the "relocation error"). Then, one can notice that in the log there is no line present informing about the version of the ublox chip, if it is found. In my case it looks like this:
[image]
(ignore the error with the missing LM75). The line showing the Ublox version is printed out only, when the ublox was actually identified, otherwise it is not. This suggests that there still seems to be a problem with the access to the uart. Probably it has to do with the way you started the daemon. If you start it by other means than via the sysd service, the daemon is not executed from user "muonuser"  which may suffer from missing privileges, such as uart access and write permission to /var/muondetector. Btw. also the folder /var/muondetector should be owned by user muonuser, not by pi.

If the problems persist, i propose to set up a clean install from scratch and compile from the current dev branch and then install the generated package. @lnies just tested this succesfully and might provide some more insight into this.

Sorry to not providing you more support at the moment. As already mentioned, the armv6-thing is quite a new endeavour for us, too.

cheers
hgz

marvin5300, 2022-01-19:
> It still looks like an incompatible ABI probably coming from the fact that the binary was linked against a different version of Qt than what is present on the image (at least i think is true for the [...]

I agree, using a clean raspberry pi OS and then installing the muonpi package which is compiled for armv6 will probably work. We should write a disclaimer that the current image will not work with raspberry pi zero.

lnies, 2022-02-02:
@CodingBobby, did you resolve the issue?

CodingBobby, 2022-02-02:
Sorry that I didn't give further updates on this. Yes, I recently managed to build a working package on a fresh Raspberry Pi OS *buster* installation after several attempts on the newer *bullseye* version.

On *bullseye*, neither installing your prebuilt packages would fix the issues I had before, nor building from source would even finish successfully. I had many dependency-issues that could not be fixed, and some required packages even refused to install.

On *buster*, the general build process worked perfectly after installing the dependencies listed in your README, though I had initial issues with `qt5-default` which only worked when being installed after the other packages (which took multiple hours alone). I compiled the package from the dev-branch, commit 5c38694 was the latest at that time.

The detector functions like a charm now; thanks for all the help! Maybe the problems I had are somewhat useful for you (or other people trying to use a Pi Zero) later on.

closed 2022-02-02 by CodingBobby