MuonPi/muondetector-old-qt5-backend #67: missing file - libpigpiod_if2.so

bodscho, 2022-02-14:
after restart I get trouble by use the daemon, it doesn't start, exit code 127
so I decide to do a fresh install (of course my Raps is also new); but my problem still exist; I try to figure out what it is 
so I try look what is the status of my daemon (systemctl) again

> ● muondetector-daemon.service - muondetector-daemon - Daemon for custom muondetector board
>      Loaded: loaded (/lib/systemd/system/muondetector-daemon.service; enabled; vendor preset: enabled)
>      Active: failed (Result: exit-code) since Mon 2022-02-14 10:00:16 CET; 6min ago
>     Process: 1210 ExecStart=/usr/bin/muondetector-daemon -c (code=exited, status=127)
>    Main PID: 1210 (code=exited, status=127)
>         CPU: 13ms
> 
> Feb 14 10:00:16 raspberrypi systemd[1]: muondetector-daemon.service: Failed with result 'exit-code'.
> Feb 14 10:00:16 raspberrypi systemd[1]: muondetector-daemon.service: Scheduled restart job, restart counter is at 5.
> Feb 14 10:00:16 raspberrypi systemd[1]: Stopped muondetector-daemon - Daemon for custom muondetector board.
> Feb 14 10:00:16 raspberrypi systemd[1]: muondetector-daemon.service: Start request repeated too quickly.
> Feb 14 10:00:16 raspberrypi systemd[1]: muondetector-daemon.service: Failed with result 'exit-code'.
> Feb 14 10:00:16 raspberrypi systemd[1]: Failed to start muondetector-daemon - Daemon for custom muondetector board.

and `sudo journalctl | grep muon*`

>> Feb 14 10:09:10 raspberrypi systemd[1]: Started muondetector-daemon - Daemon for custom muondetector board.
> Feb 14 10:09:10 raspberrypi muondetector-daemon[1265]: /usr/bin/muondetector-daemon: error while loading shared libraries: libpigpiod_if2.so: cannot open shared object file: No such file or directory
> Feb 14 10:09:10 raspberrypi systemd[1]: muondetector-daemon.service: Main process exited, code=exited, status=127/n/a
> Feb 14 10:09:10 raspberrypi systemd[1]: muondetector-daemon.service: Failed with result 'exit-code'.
> Feb 14 10:09:10 raspberrypi systemd[1]: muondetector-daemon.service: Scheduled restart job, restart counter is at 1.
> Feb 14 10:09:10 raspberrypi systemd[1]: Stopped muondetector-daemon - Daemon for custom muondetector board.

so obviously there is be missing one file (or dependence) libpigpiod_if2.so, what did I do wrong?

CodingBobby, 2022-02-14:
When I installed the daemon, I also encountered an error with the exit code 127. See issue #64 where I described it, maybe that helps you already.

`libpigpiod_if2.so` looks like a file from the packages `pigpio` or `pigpiod`, so check if those are installed correctly.

What operating system did you install on your RasPi? I had problems getting the dependencies to work properly on the Debian version `bullseye` but everything worked on the older version `buster`.

bodscho, 2022-02-14:
Thank you very much. Your Issue solved my problem. so we can mark this cause as closed; if someone else miss this lib `libpigpiod_if2.so` after update of pi - goto issue #64 and follow these step

closed 2022-02-14 by bodscho