MuonPi/muondetector-old-qt5-backend #68: Cannot connect to MQTT

CodingBobby, 2022-03-02:
After having my Pi Zero offline for a day or so (to test things regarding the architecture), I can't log in to the MQTT anymore.

When checking `systemctl status muondetector-daemon.service`, this is the displayed error:
```
...
muondetector-daemon[2307]: Warning: "Could not connect to MQTT: Bad address. Trying again in 10000ms"
muondetector-daemon[2307]: Warning: Couldn't publish data
...
```
I tried to log in again manually with `muondetector-daemon` but it yields the same error.

The detector runs fine otherwise, and I can connect with the GUI, but it displays a red "MQTT-Connection" status.

Is this a temporary server-side issue and the address is literally "bad", does my Pi have network problems (it worked previously and I didn't change anything), or is it something else you are already aware of?

hangeza, 2022-03-03:
Thanks for the error report! It seems, that after the complete server restart yesterday evening (Mar 2, 8pm), some of the detectors in the network could not reconnect to the MQTT broker with this error message. 
In order to support the forensic search for the reason of this failure, it would be very helpful if all users encountering the same problem could post an error report as a comment in this thread containing 
1) the version of the muondetector-daemon 
2) how you installed it: via standard apt install from the muonpi repo or compiling from source. In the latter case, please indicate which branch of the git repo you were compiling from.
3) specify the architecture and the RPi version as well as the version of the OS (output of `uname -a` and `cat /etc/os-release`)
4) append the log output of muondetector-daemon since last restart (e.g. with `sudo journalctl -u muondetector-daemon.service`)

All these information are much appreciated and will be helpful in encircling the failure.
Thanks
hgz

hangeza, 2022-03-04:
Problem solved! After the server restart, the floating IP was not automatically reassigned to this server and thus the newer muondetector-daemon versions which resolve the hostname data.muonpi.org to the floating IP (instead of specifying a fixed IP of the server in older versions) did not get connected to the MQTT broker. After resetting the IP assignment on the server, now all missing detectors have connection again. 
In short: this issue was caused by a server-side problem and not daemon-related

closed 2022-03-04 by hangeza