MuonPi/muondetector-old-qt5-backend #74: Unreadable labels on macOS "dark theme"

CodingBobby, 2022-05-02:
When setting the native system theme of macOS (10.14) to "dark", the GUI does automatically adapt the colours, which works fine, except in places where the background-colour is fixed to white. As the text-colour changes to white for the dark theme, the labels of plot axes render unreadable.

Here are two screenshots of the GUI (version 2.0.2 installed from [this release](https://github.com/MuonPi/muondetector/releases/tag/v2.0.2)) comparing the macOS themes "bright" and "dark":

[image]

[image]

A solution would be to fix the text-colour to black wherever the background-colour is fixed to a bright colour (notice that the boxes for time pulse and MQTT connection statuses are also barely readable). This wouldn't affect the bright theme or any other system that has no native dark theme.

hangeza, 2022-05-02:
The issue seems to be fixed for the time being with PR #75. However, it would be nice if also the Qwt widgets would follow the system-wide applied theme and switch to e.g. dark background when dark theme is selected. Is there any chance that this could be realized?

hangeza, 2022-05-03:
With commit #86c8a030fb6de14b89771c7f14f35fd4b6178b76 all widgets should now adjust automatically to the currently selected color theme. Tested under Linux Mint 20.3 (Una) and MATE desktop. Plz. reopen issue in case there are still some problems with this in your environment.

closed 2022-05-03 by hangeza